opendata-stuttgart/devices.sensor.community #2: Center map based on selected country

DeeKey, 2019-04-06:
I hardware section of sensor registration there is a map. currently it is in some dummy position.
Would be nice to center it on the capital of the chosen country for more user convenience.

ricki-z, 2019-04-06:
Link to map goes to world map. But there is a link now behind every sensor that opens a map view with the sensor in the center.

Informatic, 2019-04-07:
I think whole address/location input needs some rework by someone comfortable with "frontend" development/JavaScript and UX. We've only added "Address lookup" button as this was a simple solution that wouldn't disturb anyone, in case they don't actually want to use it (sometimes address lookup returns rather random results)

ricki-z, 2019-04-07:
@Informatic that is another "improvement". Deekey meant the map link on the sensor data page.
And that was changed, the first link to the whole world view and additional links for each sensor.

ricki-z, 2019-04-07:
@DeeKey Is the actual solution okay?

ricki-z, 2019-05-07:
@DeeKey can we close this?
The first map link can't be made language dependent. There are many countries with the same language.
But the link after the sensor ID should be okay for this.

ricki-z, 2019-11-15:
@DeeKey let's talk about closing issues ;-)
Is this solved?

DeeKey, 2019-12-20:
Yes :+1: >

closed 2019-12-26 by ricki-z