opendata-stuttgart/devices.sensor.community #9: Add language override dropdown

Informatic, 2019-04-07:
Currently selected locale/language can be overriden using `?lang=xx` request argument. This setting is stored in user session, and can be cleared using `?lang=` (empty argument)
https://github.com/opendata-stuttgart/meine-luftdaten-info/blob/0b45bdab7fcc44c01938760d2882ac7f073afb74/webapp/babel/__init__.py#L15-L20

This functionality should be exposed in user interface, for example in upper right corner, or in footer.

ricki-z, 2019-04-07:
This function sets session language to empty string if 'lang' argument is empty.

ricki-z, 2019-04-07:
But the language selection is added. Needs some css make over.

Informatic, 2019-04-07:
Yeah, and `session['lang'] == ''` will effectively make line 19/20 fall back to automatic language selection.

ricki-z, 2019-04-07:
I have checked this. 'lang' isn't set to automatic language. In the template layout.html there is an 'if' to set a class for active and inactive languages.
The following version sets all to inactive:
```
{% for language in AVAILABLE_LANGUAGES.items() %}
	{% if CURRENT_LANGUAGE == language[0] %}
		<li class="lang_active"><a name="language">{{ language[1] }}</a></li>
	{% else %}
		<li class="lang_inactive"><a href="?lang={{ language[0] }}">{{ language[1] }}</a></li>
	{%  endif %}
{% endfor %}
```
While this is working (BEST_MATCH_LANGUAGE is the best matching language):
```
{% for language in AVAILABLE_LANGUAGES.items() %}
	{% if CURRENT_LANGUAGE == language[0] or (CURRENT_LANGUAGE == '' and BEST_MATCH_LANGUAGE == language[0]) %}
		<li class="lang_active"><a name="language">{{ language[1] }}</a></li>
	{% else %}
		<li class="lang_inactive"><a href="?lang={{ language[0] }}">{{ language[1] }}</a></li>
	{%  endif %}
{% endfor %}
```

ricki-z, 2019-04-07:
For a test:
Set language by calling url with nonempty lang parameter. Then set the language with empty parameter. Call a different page by clicking one of the menu links.

Informatic, 2019-04-07:
Code in `webapp/common/context_processor.py` was wrong all along (especially `CURRENT_LANGUAGE` you are using above).

This has been fixed in PR #11 by properly using `flask_babel.get_locale` when rendering the dropdown.

This is the behaviour with PR #11 applied, which I assume is valid:
```
atuin\informatic ~ $ http --session /tmp/test http://localhost:5000 | grep -i 'nav-link' | head -1 # defaults to english
                        <a href="/" class="nav-link">Home</a>
atuin\informatic ~ $ http --session /tmp/test http://localhost:5000/?lang=pl | grep -i 'nav-link' | head -1
                        <a href="/" class="nav-link">Strona główna</a>
atuin\informatic ~ $ http --session /tmp/test http://localhost:5000/ | grep -i 'nav-link' | head -1
                        <a href="/" class="nav-link">Strona główna</a>
atuin\informatic ~ $ http --session /tmp/test http://localhost:5000/?lang= | grep -i 'nav-link' | head -1
                        <a href="/" class="nav-link">Home</a>
atuin\informatic ~ $ http --session /tmp/test http://localhost:5000/ | grep -i 'nav-link' | head -1
                        <a href="/" class="nav-link">Home</a>
atuin\informatic ~ $ http --session /tmp/test http://localhost:5000/ Accept-Language:pl | grep -i 'nav-link' | head -1
                        <a href="/" class="nav-link">Strona główna</a>
atuin\informatic ~ $ http --session /tmp/test http://localhost:5000/ Accept-Language:en | grep -i 'nav-link' | head -1
                        <a href="/" class="nav-link">Home</a>
atuin\informatic ~ $ http --session /tmp/test http://localhost:5000/?lang=en Accept-Language:pl | grep -i 'nav-link' | head -1
                        <a href="/" class="nav-link">Home</a>
atuin\informatic ~ $ http --session /tmp/test http://localhost:5000/ Accept-Language:pl | grep -i 'nav-link' | head -1
                        <a href="/" class="nav-link">Home</a>
```

ricki-z, 2019-04-07:
Seems to work.

closed 2019-04-07 by ricki-z