opendata-stuttgart/devices.sensor.community #22: flash message 'This sensor ID is already registered.' istn't translated by babel

ricki-z, 2019-04-19:


ricki-z, 2019-05-05:
There are some more places where the translation doesn't work, i.e. "Email Address" and "Password" on the user registration page.

closed 2019-05-05 by ricki-z

Informatic, 2019-05-05:
Can't this be translated via `flask_security.po` file in `translations/*`? (didn't check this yet, just asking if anyone tried this)

ricki-z, 2019-05-05:
if i "compile" the flask_security.po file then the text is translated. Is there any reason for a second.mo file?

Informatic, 2019-05-05:
Yes. The idea here is that this is flask-security module translations. In theory we should be able to just pull these from https://github.com/mattupstate/flask-security/tree/develop/flask_security/translations (and also contribute our translations when appropriate) In theory these should be automatically loaded, but *something* is broken in our codebase (presumably a conflict between Flask-Babel used by wtforms or some other module and Flask-BabelEx used by Flask-Security...) so that we need to have these "external" translations pulled into our local `translations/` directory. I've spent some time trying to debug this original issue, but couldn't figure it out without forking either of these projects... :/

PR #30 (2019-05-05): Use flask_babelex instead of flask_babel
Mixed use of flask_babel & flask_babelex sadly breaks translations randomly. This PR replaces most uses of `flask_babel` with `flask_babelex` so that everything should work fine now.

Fixes #22