opendata-stuttgart/devices.sensor.community #23: "deleting" sensors

ricki-z, 2019-04-20:
To "delete" a sensor we could set it to inactive and add a prefix "deleted-" to the email field in the API database. This will take the sensor from the user list of sensors, but we could restore it.

imaluschi, 2020-09-05:
Sounds good, nevertheless i would appreciate an option to delete a Sensor permanently. Posibly "Delete from List" and "Delete permanently"?
When doing a rework on a Sound-Sensor, i destroyed the USB-Port of an ESP8266. While trying to repair it, the ESP got broken - strangely i can still upload a program, but output is crap. So in my case, one registered Sensor UID is gone for ever.
Data is saved (export from Grafana), so the data could also be deletet (saves data-space on your servers).
Maybe it would be possible to add a button "Export Data as CSV" when a User wants to delete a Sensor permanetly?

ProBackup-nl, 2021-03-17:
Registering a device and making a **typo**, or just entering the **wrong ESP identifier**, makes you want to delete your registration.

This happens with registrations, that have **no measurement data recorded**.

Those device should be able to be self deleted. This helps the rightful owner of the ID to register their device in future. Thus related to issue: #60