opendata-stuttgart/devices.sensor.community #26: Countries with a comma in the pycountry list aren't translated

ricki-z, 2019-05-03:
Original version produced an array with 3 fields (country code, part 1 of name, part 2 of name).
Replaced "," with " - " but now it's not translated anymore.

Informatic, 2019-05-05:
Running `.replace` is done before pushing a string through translation framework. Moving this outside wouldn't work either, as this would make lazy translation impossible (thus everything would still be passed around as plain `str` in default language)

`.replace` should be done in `__init__` of `SensorLocationForm` (post-translation). I can create a PR for that.

What is a reason for that replacement, though? This should rather be fixed in upstream project, but even so, comma is a very common and accepted delimiter in such situations.

ricki-z, 2019-05-05:
The "function" that is generating the list for the select returns i.e. for Iran:
IR, Iran, Islamic Republic of
This results in a select option with Text Iran and a value of "Islamic republic of", which is then denied as an unavailable country.
I don't know why python is creating an array with 3 cells from this.
Comma is a common delimiter for a list of countries, but its not working for meine.luftdaten.info . So WE have to find a solution for this.

Informatic, 2019-05-05:
OK, looking into this now, and this seems *very* strange.
HTML is rendered as:
`<option value="IR">Iran, Islamic Republic of</option>`
but then parsed into:
`<option value=" Islamic Republic of">Iran</option>`

Informatic, 2019-05-05:
https://github.com/opendata-stuttgart/meine-luftdaten-info/blob/develop/webapp/templates/sensor-settings-form.html#L48-L75 Seems like this piece of JS breaks that form field.

Informatic, 2019-05-05:
Sorting should be easily doable on python side by the way... (I assume select options are sorted by country code by default, and we want the select box to be sorted based on human-readable country name?)
I'll fix this tonight.

ricki-z, 2019-05-05:
Thanks for the tipp with the javascript. I forgot about this. The javascript is working now as expected even with the "original" country names.

closed 2019-05-07 by ricki-z