opendata-stuttgart/devices.sensor.community #33: Sensor-Konfiguration auf https://meine.luftdaten.info

thjean, 2019-05-05:
Mit Freude habe ich gelesen, dass die Registrierung und Konfiguration der Sensoren auf https://meine.luftdaten.info nun selbst vorgenommen werden kann - vielen Dank dafür!

Interessanterweise sehe ich in den Einstellungen der Sensortypen auch den SHT31/35, den ich in meiner Haussensorik gerne verwende, da er gegenüber dem BME280 z.B. einige Vorteile im Aussen-Bereich hat (die Luftfeuchte geht nicht auf 100% und der Sensor kann bei Bedarf beheizt und getrocknet werden. Hinzu kommt, dass insbesondere der SHT35 deutlich genauer misst als der BME280, was aber für diese Anwendung nicht relevant ist). 
Mit der Auswahl des SHT3x und der PIN 7 stehe ich aber vor dem Problem, dass die Konfiguration insgesamt nur zwei Sensortypen umfasst dass ich damit zwar Temperatur und Feuchte als SHT3x kennzeichnen kann, es aber nicht mehr möglich ist, neben dem Feinstaubwert auch noch den Luftdruckwert eines BME280 zu senden. Beim Versuch, zusätzlich zu Temp/Feuchte mit PIN=7 und Feinstaub noch den Luftdruck mit der nicht eingetragenen PIN=11 zu senden bekomme ich diese Antwort vom Server, da offensichtlich nur die zwei in der Konfiguration eingetragenen PINs akzeptiert werden: 
````
HTTP/1.1 400 Bad Request
Date: Sun, 05 May 2019 11:54:11 GMT
Server: Apache/2
Allow: POST, OPTIONS
Connection: close
Transfer-Encoding: chunked
Content-Type: application/json

21
["sensor could not be selected."]
0
````

Testweise habe ich unter Verzicht auf Feinstaubwerte als ersten Sensortyp den SHT31 eingetragen und als zweiten Typ den BME280. Damit können Temp/Feuchte mit PIN=7 und Luftdruck mit PIN=11 gepostet werden, die Werte werden auch vom Server akzeptiert. Auf maps.luftdaten.info werden mit dieser Konfiguration die mit PIN=7 gepostete Temp/Feuchte Werte nicht dargestellt, sondern nur der mit PIN=11 gepostete Luftdruck.

Mir ist klar, dass es den Use-Case mit der (aktuellen) airrohr-firmware nicht gibt, für Anwender mit eigener Firmware wäre es jedoch geschickt, wenn insgesamt mehr als zwei Sensortypen konfigurierbar wären, so dass für Temperatur, Feuchte und Luftdruck unterschiedliche Sensortypen akzeptiert würden.

ricki-z, 2019-05-05:
Zunächst habe ich den Issue nach meine-luftdaten-info verschoben, da passt er meines Erachtens nach besser.
Ich lasse den Issue hier stehen und werde mir das bei Gelegenheit genauer anschauen. Die Karte sollte eigentlich beide Sensoren SHT31 und BME280 anzeigen. Beide haben ja jeweils eine eigene ID.
Was die Werte angeht: Wir zeigen keine Nachkommastellen an, weil die "Genauigkeit" aller Werte nicht wirklich mehr zulässt. Wir sind hier schon fast angegangen worden, wie wir uns getrauen können, Nachkommastellen bei den PM-Werten anzuzeigen. Damals hatten wir auf Grund der Mittelwertbildung eine Nachkommastelle bei den Werten angezeigt.
Bei der Temperatur sollten wir das aber ändern können.

thjean, 2019-05-06:
Hallo,
Ich kann auch nach weiteren Versuchen bestätigen, dass mit "X-Pin: 7" gesendete Werte auf der Map nicht angezeigt werden, obwohl sie von der API mit ````HTTP/1.1 201 Created```` akzeptiert werden. 
Das ist aber überhaupt kein Problem, wenn die Pin nicht weiter ausgewertet wird,  sende ich mit "X-Pin: 11" eine Mischung aus SHT31 und BME280 Daten. 
Wird mit "X-Pin:11 " nur der Luftdruckwert gesendet (also ohne Temperatur und Feuchte), dann wird auch dieser Wert von der API mit ````HTTP/1.1 201 Created```` akzeptiert, aber auf der Map nicht dargestellt. In diesem Fall wird weiterhin der letzte zuvor gesendete Luftdruckwert aus einem "X-Pin:11" Paket mit allen drei Werten dargestellt. 
Zu den Nachkommastellen bin ich der gleichen Meinung: Angesichts der Genauigkeit der Sensoren/Messungen ergeben Nachkommastellen keinen Sinn, bestenfalls kann bei der Temperatur eine Nachkommastelle die Veränderung zeigen. Zumindest bei dem üblichen - und sehr anwenderfreundlichen - Aufbau mit HT-Rohren sind die Belüftungsverhältnisse nicht ideal, so dass  Temperatur und Luftfeuchte deutlich abweichen von den Werten, die an gleicher Stelle bei besserer Belüftung gemessen werden. Von den vielen DHT, deren Luftfeuchte nahe 100% hängt, gar nicht zu reden.
Ich möchte nur anregen, die Werte in der Tabelle zu runden (egal ob kaufmännisch oder mathematisch) und nicht die Nachkommastellen abzuschneiden und dies auch einheitlch für alle Phänomenen zu tun.
Zum Schluss noch ein großes Lob für die tolle Arbeit, ich kann mir ungefähr vorstellen, welches persönliche Engagement dahinter stehen muss. Insofern bitte meine Anmerkungen nicht als Kritik, sondern als nice-to-have Verbesserungen ansehen.

ricki-z, 2019-05-06:
Jetzt sollten die SHTs angezeigt werden. Diese waren noch nicht auf der Karte als Temperatursensoren eingetragen.

thjean, 2019-05-06:
Hallo,
Danke für den Fix. Die Anzeige der SHT-Werte in der Map klappt leider trotzdem noch nicht. 
Ich vermute, dass woanders noch ein Wurm sitzt, denn die API liefert mir nach dem Posten der SHT (Test-)Werte die Sensor-ID 20939 für die Feinstaubwerte zurück statt der ID 20940 für Temperatur, Feuchte und Luftdruck. Dies passiert sowohl bei der Konfiguration "SDS011 X-Pin:1" und "SHT31 X-Pin:7" als auch mit der Konfiguration "SHT31 X-Pin:7" und "BME280 X-Pin:11".
 
Gesendete Testdaten:

```` POST /v1/push-sensor-data/ HTTP/1.1
Host: api.luftdaten.info
Content-Type: application/json
X-PIN: 7
X-Sensor: esp32-12078136
Content-Length: 143
Connection: close

{"software_version": "2.98.B2.10", "sensordatavalues":[{"value_type":"temperature","value":"11.11"},{"value_type":"humidity","value":"55.55"}]}
````
API-Antwort: 
````
HTTP/1.1 201 Created
Date: Mon, 06 May 2019 12:34:44 GMT
Server: Apache/2
Allow: POST, OPTIONS
Connection: close
Transfer-Encoding: chunked
Content-Type: application/json

82
{"sensor":20939,"timestamp":"2019-05-06T12:34:44.436043","sensordatavalues":[{"sensordata":3554077181},{"sensordata":3554077181}]}
0
````

ricki-z, 2019-05-06:
Ich habe jetzt noch den SHT31 zum Eintrag hizugefügt (ID ist 25735).
Damit sind jetzt alle 3 Einträge vorhanden:
Pin 1: SDS011 (20939)
Pin 11: BME280 (20940)
Pin 7: SHT31 (25735)

thjean, 2019-05-06:
Oh, danke!
Dann teste ich mal weiter und berichte.

thjean, 2019-05-06:
Es ist wie verhext... Die API nimmt die Daten für alle drei Pins entgegen, ober die Map will den SHT mit ID 25735 einfach nicht anzeigen, egal ob ich nur diese eine Pin sende oder zwei oder alle drei :)

ricki-z, 2019-05-06:
Unter http://api.luftdaten.info/v1/sensor/25735/ kann man die Daten sehen. Irgendwo beim Aufbereiten der Daten für die Karte scheinen die Daten "verloren" zu gehen. Ich schau mir das nochmal an.

ricki-z, 2019-05-07:
Jetzt sollten alle 3 Komponenten auf der Karte erscheinen.

thjean, 2019-05-07:
Ja, ich kann die Werte aller drei Sensoren seit einigen Stunden auf der Karte sehen - aber nur in der Tabelle, der Graph für den SHT31 ist leer.

ricki-z, 2019-05-07:
Dort sollten jetzt auch nach und nach Werte kommen. Die von heute spiele ich noch ein, mehr bräuchte recht lang.

thjean, 2019-05-07:
Danke nochmal für die Extra Mühe, bitte keinen Aufwand machen. Es reicht doch vollkommen, wenn die Werte künftig im Graph erscheinen.

ricki-z, 2019-05-07:
Ab und zu bleibt die Influx-Datenbank oder Grafana hängen (die Kombi erzeugt die Grafiken). Dann muss ich die Werte manchmal nachimportieren. Insofern ist das gleich ein Testlauf, das dort auch die SHTs importiert werden. Und damit ist nur der Server beschäftigt, ich sitze nur "daneben" und warte ;-)

thjean, 2019-05-07:
Prima, die SHT-Werte kommen jetzt, ich sehe dann mal morgen nach, ob auch die 24h floating Werte einlaufen. 
Ich traue mich schon fast gar nicht mehr, noch was zu schreiben, aber  .... im Temperatur-Graph heißt der Sensor "SHT22". Ist das Ganze eine Sonderausgabe für mich - das hätte ich gar nicht verdient :)

molchi75, 2019-06-05:
Hallo,

ich habe ein ähnliches Problem. Mein raspi lädt die Daten des sds011 und des dht22 alle 150 Sekunden hoch, unter http://api.luftdaten.info/v1/sensor/26429/ http://api.luftdaten.info/v1/sensor/26430/ sind die Daten auch zu sehen. In der Map jedoch und in Grafana werden sie nicht angezeigt. 
Für Hilfe wäre ich dankbar :)

ricki-z, 2019-06-05:
Im gesendeten JSON kommt z.B. für die PM10-Werte als 'value' "{'value': '3.90'}". Hier darf nur der Wert kommen. Es ist also nicht wirklich ein "ähnliches Problem".

molchi75, 2019-06-05:
Danke für die schnelle Antwort, habe meinen Fehler im Programm gefunden.