opendata-stuttgart/devices.sensor.community #38: Some issue with the French translation

MagTun, 2019-07-25:
There are some mistakes on the French translations but I can't find the original strings in the file: https://github.com/opendata-stuttgart/meine-luftdaten-info/blob/develop/translations/fr/LC_MESSAGES/messages.po

[image]
[image]

I copy them here so it will be easier for you to copy them:  
Adresse Email  
Mot de passe  
Garder ma session active  
Se connecter    
Cela peut prendre jusqu'à 30 minutes pour recevoir l'email d'enregistrement. 
Si vous n'avez pas reçu d'email, merci de vérifier de la catégorie Spams.   
S'enregistrer  

Also, I have created a pull request on the wrong reposetory for the French translation: https://github.com/Informatic/meine-luftdaten-info/pull/1 
Is there a easy way to send the pull over to you or should I create a new pull request from scratch (I have seen that all the strings have been translated but some could be improved).

ricki-z, 2019-07-26:
Hello,
the mentioned strings are part of the flask security extention which needs to be "translated" outside the normal source tree.
I will try to insert your translations for this part.
To your other question:
The easiest way would be to copy the changed files to a location outside your local repo copy. Then delete your existing fork (from Informatic) and fork our version. Use this fork as your local work copy and copy back the changed files. After pushing these changes to your forked version you should be able to create a pull request against our repo.

ricki-z, 2019-07-26:
After a second look I have seen that these messages are located in the flask_security.po file. So you could change this file and the strings should be shown in French (after we copied it to the server ;-) ).

MagTun, 2019-07-26:
OK, it's done for `flask_security.po` cf https://github.com/opendata-stuttgart/meine-luftdaten-info/pull/39
I will do the update of `messages.po` later.

closed 2019-07-26 by MagTun