opendata-stuttgart/devices.sensor.community #40: A question is missing in the registration form about barbecue and vegetal waste burning

MagTun, 2019-08-13:
Your registration form mentions stoves and fireplaces ( line 80 and 92 of [form.py](https://github.com/opendata-stuttgart/meine-luftdaten-info/blob/develop/webapp/personal/forms.py)), but you don't mention other wood-burning sources of PM like barbecue and burning vegetal waste in one's garden. These event creates a high peak of PM during a few hours, and/or contribute to the accumulation of particles during specific weather. 

I live in a small French city, which has no industry, low traffic, no highway... Most of the pollution comes from people fires: during spring and summer, it comes from barbecues, people burning their garden waste or farmers burning vegetal waste. These come on top of to the chimney smoke in winter, so it smells like burning several times per day, during the whole year.

[image]

To help interpret the data, you could simply add this missing question:
**"Do people burn vegetal waste in your area or barbecue ?  1 = very little, 10  very much."**

ricki-z, 2019-08-27:
Something like this should be added to the description field.
Such very subjective questions were removed from the form. For different people the same neighbors might give very different values.
Example: In the past I got many messages where people told us that they live on a street with high traffic with around 5.000 cars per day. I live on a street with 18.000 cars per day and near a crossing with more than 150.000 cars per day. For me a street with 5.000 or even 10.000 cars would be a street with low traffic.

MagTun, 2019-08-27:
I understand your point but I think we can find a way to measure the frequency/intensity of waste burning and barbecue in a (quite) objective way. For instance, a question like "How many garden waste fire are there in your neighborhood per month?" will generate quite pretty objective answers (the answers will be even more objective than the ones you'll get for the form question: "How much industrial activity is there within a 100m radius? No office space, but potential fine dust producers 1 = very little, 10 = very much"). The smell of vegetals burning isn't hard to smell, and not so subjective either.

Or you could simply append a question like the one suggested above to this form question "'How many private stoves or fireplaces are within a 100m radius?". Both questions fit to the current description "Does it smell very much like such smoke in your area?" and complete each other. 

Garden waste fire are a major source of PM so it would help interpret the data to have such information.

ricki-z, 2019-08-28:
If you take a look at the actual version of the registration form you will see that all those questions where removed. We had a request to remove these from an organization that would like to use the data for scientific research. And they told us that the answers to these questions couldn't be used because of their subjectivity.

MagTun, 2019-08-30:
Sorry, I didn't look at the registration form on the website, only at the [form.py](https://github.com/opendata-stuttgart/meine-luftdaten-info/blob/develop/webapp/personal/forms.py), assuming it is used as the source for the actual version.  

I believe it's a mistake to have removed the questions at the request of the organization. We are in a learning process with Citizen Science so not everything can be perfect. If we simply remove all the obstacles then we will never learn. Science improves by trial and error and I think that keeping the questions and finding ways to make them less subjective is how we can progress. In this learning process, unprecise data are better than no data. Luftdaten is one of the biggest Citizen Science projects and the decisions Luftdaten takes could have a big impact on the future of this new way of doing science.

In brief, I think it's important to keep the questions even if they are not objective enough for the moment. If the organization thinks these data are not reliable, they can simply ignore them.

ricki-z, 2019-08-31:
May I forward you to Pierre Dornier from transportenvironment.org for this?
Most of these (not objective) informations could be retrieved from other sources. And in some cases it seems to be better if you don't know about it to work with the data uninfluenced. There were some works at a data science challenge earlier this year. They had only the data we publish in our archive and used additional data sources. Take a look at this at https://link.springer.com/epdf/10.1007/s13222-019-00322-x?author_access_token=betuIj4zoQjwUNS08eLCkfe4RwlQNchNByi7wbcMAY4XSJGLy5YKWOWvMAACsUy2TQdfdJhs8AaI5GEJj8J50cU0aC0pwDYVthDGrmEwVsQ0SHHI6zSZH0K0Zc-5IDJ7OzJyGKHhy_f-XR_2hztRFg%3D%3D

MagTun, 2019-09-03:
I read the studies, it's great to see people using the luftdaten data in such a way! I still think it's a mistake to have remove the questions from the form, I will contact Pierre Dornier to understand his point.