opendata-stuttgart/devices.sensor.community #45: Moving to another town: what is the recommended procedure to keep data from before/after separated

MagTun, 2019-11-14:
When moving to a new place, the data from before and after should be kept completely separated. If I understood right, the sensor are identified by a ID so is there a way to get a new ID?

ricki-z, 2019-11-15:
You don't need a new ID. There is a location ID in all JSON data files and also in the archive files. This location ID will change if you change the location of your sensor at my.luftdaten.info.

MagTun, 2019-11-15:
Super. Just one last question: will  the madavi graph be "reset" or will there be a continuity with the data from the previous location still displayed?

ricki-z, 2019-11-15:
The madavi graph will be continued. It only depends on the ESP8266 id.

MagTun, 2019-11-17:
Ah ok, so I will change my esp8266 so I don't get my previous data on my new graph. Thanks!

closed 2019-11-17 by MagTun