opendata-stuttgart/devices.sensor.community #56: permanent server overload?

michapr, 2020-03-31:
I get most time no data from server, only the empty map.
A self registration is not possible because of server timeout.

Any issues with server overload?

Thanks,
Michael

haup, 2020-04-01:
I am also getting timeouts for my sensor probably since a couple of days. It seemed yesterday a bit better (api answered, but sensor data wasnt visible via webgui) , but this morning again no api response at all.

ricki-z, 2020-04-01:
We are working on this. As the traffic has increased massively in some data centers the internal networks are sometimes overloaded.
For the moment it seems like we could solve this for the registration page, but not for our main database.

closed 2020-05-22 by ricki-z