opendata-stuttgart/devices.sensor.community #60: This sensor ID is already registered.

felixbro, 2020-05-10:
Hello! ich würde gerne meinen Sensor in meine.luftdaten eintragen, werde aber dort seit einigen Tagen mit dem Hinweis: "This sensor ID is already registered." konfrontiert.
Was kann ich denn da genau machen? Habe zur Zeit 3 DNMS laufen und konnte nur zwei registrieren.
____________
Hello! I would like to enter my sensor in my air data, but have been there for a few days with the note: "This sensor ID is already registered." faced.
What exactly can I do there? I currently have 3 DNMS running and could only register two.

ricki-z, 2020-05-12:
Please double check the chipID first (on sensor and at registration). Sometimes it's only a typo.
The only quick fix, if the problem persists, is to replace the NodeMCU to get a new chipID.

Jaxom99, 2020-07-13:
This also happen when you were registered "manually" an the early stages of the project ("luftdaten"), which is my case. Can we track progress/solutions on the matter in this ticket ?

hyperion-cs, 2024-04-25:
Same thing with my sensor 5813634.
```
 This sensor ID is already registered.
```

ricki-z, 2024-04-28:
Please don't use this thread for requests anymore. There is a thread at https://forum.sensor.community/ for this which we look at more often. And it should avoid double work.