opendata-stuttgart/devices.sensor.community #62: SDS011 & BME280 - how to add SHT35?

sumerland, 2020-06-15:
Hi,

ich hoffe, ich bin hier an der richtigen Stelle mit dem folgenden Problem:

Ich habe heute zu dem bisher laufenden Paar SDS011 & BME280 einen SHT35 hinzugefügt und in der Konfiguration aktiviert (Firmware: NRZ-2020-130-B6/DE). Ich lade zu madavi, sensor.community und opensensemap hoch. In den aktuellen Werten wird der SHT35 angezeigt. Auf maps.sensors.community werden aber nur Feintaub und die Werte des BME280 angeboten. Auf https://devices.sensor.community/sensors sehe ich keinen Weg, einen dritten Sensor anzumelden. Im Debug Fenster taucht folgende Meldung auf:

```
## Sending to sensor.community - SHT3x
Request failed with error: 400
Details:["sensor could not be selected."]
```

Was mache ich falsch?

ricki-z, 2020-06-18:
Hallo,
leider ist das Hinzufügen eines dritten Sensors über devices.sensor.community aktuell noch nicht möglich. Sende mir bitte eine Mail an tech (at) sensor.community mit der chipID des Sensors, dann füge ich den SHT35 hinzu.

ceage, 2020-08-24:
Hallo @ricki-z, gilt das auch für einen Lärmsensor? Ich habe diesen nach Anleitung gebaut und aktiviert und würde ihn auch gerne in die Karte integrieren. Danke!