opendata-stuttgart/devices.sensor.community #70: Please update English strings in PO files for translated languages.

DeeKey, 2021-01-05:
Some English strings were updated (mostly due to rebranding to sensor.community).
But translated PO files were not updated accordingly. Because of that some translation strings are not working and there are "holes" of unstranslated English sentences in all languages.

ricki-z, 2021-01-05:
I'm checking this.

ricki-z, 2021-01-05:
Someone used “ (left double quotation mark, UTF-8 xE2809C) instead of " (quotation mark, UTF-8 x22) as string end quote ...
And a 's' (for string) was missing after a variable to replace.

If the 'internal server error' is shown now then it's now caused by another problem.

DeeKey, 2021-01-05:
No, I was talking about different issues.
Few examples:

General information can be found at Sensor.Community - formerly known as Luftdaten.info, as well as the building instructions. Sensor data can be found on the fine dust map or be downloaded as CSVs. 
 
Donate for Sensor.Community now on Betterplace.org

ricki-z, 2021-04-07:
English strings are updated.

closed 2021-04-07 by ricki-z