opendata-stuttgart/devices.sensor.community #76: Formatting of Air pressure is wrong

hubbergit, 2021-06-06:
https://devices.sensor.community/sensors/<12345>/data shows

10.18 hPa for air pressure and 10.24 hPa for air pressure at sea level.

That should be 1018 hPa and 1024 hPa instead.

pjgueno, 2021-06-06:
Can we have the sensor ID because my sensor display correct values:
[image]

What kind of set up do you have ? Which firmware version ?

hubbergit, 2021-06-07:
Thanks for your response. I am running ESPHome instead of the default firmware. I understand that the mistake may be on my sending side. I will double check this and report back here later today.

hubbergit, 2021-06-07:
Some background: I am measuring the particles with a SDS011 and the temperature, humidity and pressure with a BME280 with a NodeMCU ESP8266. The firmware of the ESP8266 is ESPHome. The raw measurement data is handed over to Home-Assistant from where it is send to the sensor.community.

I am using the Web API like this:

URL: https://api.luftdaten.info/v1/push-sensor-data/
METHOD: POST
HTTP-HEADER: {
   "X-Sensor": <sensor_uid>,
   "X-PIN": "11"
}
HTTP-BODY: {
   "sensordatavalues":[
       { "value_type": "temperature", "value": 21.3 },
       { "value_type": "humidity", "value": 48.1 },
       { "value_type": "pressure", "value": 1018.5 }
   ]
}

I get a status code 201, which indicates that the request was accepted and the record was created. However the value is displayed as described in the ticket.

@pjgueno Any ideas?

ricki-z, 2021-06-07:
The API expects the pressure data in Pascal (as most sensors indicate). So sending us the values in Pascal (hPa * 100) should solve this.

hubbergit, 2021-06-08:
That hint helped me to understand the problem. There was a bug in the "Luftdaten"-Integration of Home-Assistant, which funnily enough was recently fixed by me. After upgrading to the latest version of Home-Assistant, I had to adjust my code to account for the bugfix.

Everything is displayed as expected now.

Thank you very much for your help.

closed 2021-06-08 by hubbergit