opendata-stuttgart/devices.sensor.community #77: Add sensors to existing configuration

dupuysylvain, 2021-06-22:
Hello,

I hace created a new config with 1 sensor.

[image]

i want to add news sensors to the configuration but i can't : when i want to edit, we can't add sensor.

So i think to remove and recreate but when i try to remove but :

[image]

so if i deleted, i can't recreate for that chip...

pjgueno, 2021-06-22:
Add a new station or a new sensor on the station ? It is not the same at all...
Read here in French:
https://sensor.community/fr/sensors/airrohr#Configuration
https://sensor.community/fr/sensors/airrohr#Enregistrement
You probably have to access your sensor in the network.

or can't you see this button in devices.sensor.community ?

[image]

dupuysylvain, 2021-06-22:
> Add a new station or a new sensor on the station ? It is not the same at all...
> Read here in French:
> https://sensor.community/fr/sensors/airrohr#Configuration [...]

Hey, thank for your reply,

i see this button ("register new sensor") but this is not what i need.

I have a board register with 1 sensor : 
[image]

and i need to add a SDS011 sensor. It's posible during a creation but no when modifying. 

Do you have a solution ?

Thanks

pjgueno, 2021-06-22:
I can't understand why there is only one sensor...

Anyway, it is what I thought. You have to connect to your station through your router: Look on the DCHP client list of your router and get the IP of the station. Something like 192.168.0.104. Then copy past in a browser.

OR

Use the search function of the flashing tool while connected on the right WiFi network (of course!) and double-click on the line.
[image]

You should reach such a page:

[image]

Then click "Configuration"

[image]

Send me a picture.

pjgueno, 2021-06-25:
OK I found the answer in our forum: https://forum.sensor.community/t/can-not-add-second-or-more-sensors-after-deleting-it-once/550/13

> Please send an email with the chipID to tech (at) sensor.community. We will add the BME280 again manually. (This was mentioned in this thread more than one time).

xek, 2024-10-22:
I see this was implemented but never published to production?