opendata-stuttgart/devices.sensor.community #88: Make address optional if geoposition is provided

LuchtwachtersDelft, 2021-10-10:
If the geoposition is entered manually and the address omitted, the form refuses to submit because the address fields are required.

What is the use-case?

- user does not want to disclose their home address, let alone have it stored in a public database
- the sensor is not installed at a proper address, for example a lamp post, in a park or in a forest

LuchtwachtersDelft, 2021-10-14:
Fantasy address is an acceptable workaround.

ricki-z, 2021-10-14:
We don't write this on the registration page as it may 'confuse' some people. The are giving such a fantasy address and are wondering why the search won't find GPS coordinates (even if we mention this explicitly, this is called 'selective reading' :wink:).