opendata-stuttgart/devices.sensor.community #89: Temperature values are not displayed on map

tobias-tengler, 2022-03-23:
My P1 and P2 values are displayed on the map on sensor.community, but not the humidity and temperature are not. Looking at the dashboard the values are present though: 
[image]
Also I get back my values through the API: https://data.sensor.community/airrohr/v1/sensor/71307/

My sensor is marked as indoors - does that make a difference?

pjgueno, 2022-03-23:
What Kind of sensors did you choose in devices.Sensor.Community ?

Please send a Screenshot.

pjgueno, 2022-03-23:
I think the scd30 is not set as RH/temp sensor in the map. Ist should be the reason.
As the co2 is not displayed on the map choose for example DHT22 in devices... It should work

tobias-tengler, 2022-03-23:
These are the sensors I currently have configured: 
[image]

They match the actual hardware.

Thanks for the suggestion about switching to DHT22 - I'll try it out :)

tobias-tengler, 2022-03-23:
After the switch to DHT22 it is indeed working as expected. Thanks for the help :)

closed 2022-03-23 by tobias-tengler